clearpathrobotics/clearpath_common #24: Unable to generate setup.bash file

AmmarW, 2023-07-28:
I have been trying to simulate the Clearpath platforms based on the Humble documentation found at;
https://docs.clearpathrobotics.com/docs/ros/tutorials/simulator/simulate

All of the guides were followed regarding the Remote Computer Setup and Clearpath Simulator. When I try to generate the setup.bash file for the robot.yaml configuration, it ends up throwing an error regarding get_model(). I believe a recent commit to the clearpath_config.py has resulted in this since a number of modules were removed but no corresponding changes were made to this repo resulting in said error. 
 
Please see the following:

`ammar@aw-asus:~/clearpath_ws$ ros2 run clearpath_generator_common generate_bash -s ~/clearpath
Traceback (most recent call last):
  File "/home/ammar/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_bash", line 46, in <module>
    main()
  File "/home/ammar/clearpath_ws/install/clearpath_generator_common/lib/clearpath_generator_common/generate_bash", line 41, in main
    bg = BashGenerator(setup_path)
  File "/home/ammar/clearpath_ws/install/clearpath_generator_common/local/lib/python3.10/dist-packages/clearpath_generator_common/common.py", line 246, in __init__
    self.platform_model = self.clearpath_config.get_model()
AttributeError: 'ClearpathConfig' object has no attribute 'get_model'
[ros2run]: Process exited with failure 1
`

roni-kreinin, 2023-07-28:
Fixed in #23.

closed 2023-07-28 by roni-kreinin

AmmarW, 2023-07-28:
Unfortunately, it is still not working. I cloned all three repos; clearpath_common, clearpath_config, clearpath_msgs just today.

Please find attached the log when running `ros2 launch clearpath_gz simulation.launch.py` which shows similar errors in multiple instances.
[launch.log](https://github.com/clearpathrobotics/clearpath_common/files/12196152/launch.log)

roni-kreinin, 2023-07-28:
#23 has not been merged yet. It will be shortly.