clearpathrobotics/clearpath_common #30: tutorial site laggin behind for hokuyo_ust10 to hokuyo_ust change.

y2d2, 2023-08-22:
**Please provide the following information:**
 - OS: Ubuntu22.04
 - ROS 2 Distro: humble
 - Built from source or installed: source 
 - Package version: 0.0.9-1jammy.20230815.024508
 - Real hardware or simulation: Not applicable

 **Issue**
When following https://docs.clearpathrobotics.com/docs/ros/installation/remote_pc/ 
`source /etc/clearpath/setup.bash` throws following error: 
```
Traceback (most recent call last):
  File "/opt/ros/humble/lib/clearpath_generator_common/generate_bash", line 46, in <module>
    main()
  File "/opt/ros/humble/lib/clearpath_generator_common/generate_bash", line 41, in main
    bg = BashGenerator(setup_path)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_common/common.py", line 246, in __init__
    self.clearpath_config = ClearpathConfig(self.config)
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/clearpath_config.py", line 97, in __init__
    super().__init__(setters, config)
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/types/config.py", line 55, in __init__
    self.config = config
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/types/config.py", line 104, in config
    self.setter(prop)(get_from_dict(value, keys))
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/clearpath_config.py", line 192, in sensors
    self._sensors.config = config
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/types/config.py", line 104, in config
    self.setter(prop)(get_from_dict(value, keys))
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/sensors.py", line 386, in lidar2d
    sensor = Lidar2D(d['model'])
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/sensors.py", line 158, in __new__
    cls.assert_model(model)
  File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/sensors.py", line 150, in assert_model
    assert model in cls.MODEL, (
AssertionError: Model 'hokuyo_ust10' must be one of: 'dict_keys(['hokuyo_ust', 'sick_lms1xx'])'
```

The robot.yaml was copied from: https://docs.clearpathrobotics.com/docs/ros/config/yaml/
where the model of the 2d lidar has not been adapted to `hokuyo_ust` and still shows `hokuyo_ust10`

By changing this value in the robot.yaml file the issue was solved. Not sure if used the wrong resource or maybe if the site is lagging behind the update of the repo. 

**additional question**
I have a real husky which has been prepared at clearpath with GPS etc. I suppose I should have a custom robot.yaml file for my robot, where can I find it? Can i find it on the husky ? 

Cheers,

y2d2, 2023-08-22:
There seem to be more fishy things going on around this hokuyo_ust. 

Not sure if I have to create a different bug report, but when I launch the simulation  `ros2 launch clearpath_gz simulation.launch.py` I get multiple errors that are related to this: 
```
root@husky:/# ros2 launch clearpath_gz simulation.launch.py
...
[generate_description-3] ImportError: cannot import name 'HokuyoUST10' from 'clearpath_config.sensors.types.lidars_2d' (/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/types/lidars_2d.py)
...
[generate_launch-4] ImportError: cannot import name 'HokuyoUST10' from 'clearpath_config.sensors.types.lidars_2d' (/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/types/lidars_2d.py)
...
``` 
There are other errors in the output, however, I am not sure if they follow because of the initial errors.

Maybe I need to build the packages from source instead of installing them?  

**full Output:** 
```
root@husky:/# ros2 launch clearpath_gz simulation.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-08-22-10-50-11-385905-husky-147
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [148]
[INFO] [parameter_bridge-2]: process started with pid [150]
[INFO] [generate_description-3]: process started with pid [153]
[generate_description-3] Traceback (most recent call last):
[generate_description-3]   File "/opt/ros/humble/lib/clearpath_generator_common/generate_description", line 36, in <module>
[generate_description-3]     from clearpath_generator_common.description.generator import DescriptionGenerator
[generate_description-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_common/description/generator.py", line 41, in <module>
[generate_description-3]     from clearpath_generator_common.description.sensors import SensorDescription
[generate_description-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_common/description/sensors.py", line 34, in <module>
[generate_description-3]     from clearpath_config.sensors.types.lidars_2d import BaseLidar2D, HokuyoUST10, SickLMS1XX
[generate_description-3] ImportError: cannot import name 'HokuyoUST10' from 'clearpath_config.sensors.types.lidars_2d' (/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/types/lidars_2d.py)
[ERROR] [generate_description-3]: process has died [pid 153, exit code 1, cmd '/opt/ros/humble/lib/clearpath_generator_common/generate_description -s /root/clearpath/ --ros-args -r __node:=generate_description'].
[INFO] [generate_launch-4]: process started with pid [158]
[generate_launch-4] Traceback (most recent call last):
[generate_launch-4]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_launch", line 36, in <module>
[generate_launch-4]     from clearpath_generator_gz.launch.generator import GzLaunchGenerator
[generate_launch-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_gz/launch/generator.py", line 41, in <module>
[generate_launch-4]     from clearpath_generator_gz.launch.sensors import SensorLaunch
[generate_launch-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_gz/launch/sensors.py", line 34, in <module>
[generate_launch-4]     from clearpath_config.sensors.types.lidars_2d import BaseLidar2D, HokuyoUST10, SickLMS1XX
[generate_launch-4] ImportError: cannot import name 'HokuyoUST10' from 'clearpath_config.sensors.types.lidars_2d' (/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/sensors/types/lidars_2d.py)
[ERROR] [generate_launch-4]: process has died [pid 158, exit code 1, cmd '/opt/ros/humble/lib/clearpath_generator_gz/generate_launch -s /root/clearpath/ --ros-args -r __node:=generate_launch'].
[INFO] [generate_param-5]: process started with pid [169]
[generate_param-5] Generated config: /root/clearpath/platform/config/control.yaml
[generate_param-5] Generated config: /root/clearpath/platform/config/imu_filter.yaml
[generate_param-5] Traceback (most recent call last):
[generate_param-5]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_param", line 46, in <module>
[generate_param-5]     main()
[generate_param-5]   File "/opt/ros/humble/lib/clearpath_generator_gz/generate_param", line 42, in main
[generate_param-5]     gpg.generate()
[generate_param-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_common/param/generator.py", line 64, in generate
[generate_param-5]     self.generate_platform()
[generate_param-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_gz/param/generator.py", line 49, in generate_platform
[generate_param-5]     platform_param.generate_parameters(use_sim_time=True)
[generate_param-5]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/clearpath_generator_common/param/platform.py", line 140, in generate_parameters
[generate_param-5]     imus = self.clearpath_config.sensors.get_all_imu()
[generate_param-5]   File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/clearpath_config.py", line 187, in sensors
[generate_param-5]     self._sensors.config[self.SENSORS])
[generate_param-5]   File "/opt/ros/humble/lib/python3.10/site-packages/clearpath_config/common/types/config.py", line 87, in config
[generate_param-5]     self.getter(prop)()
[... cut ...]
[ruby $(which ign) gazebo-1] [GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[ERROR] [ruby $(which ign) gazebo-1]: process[ruby $(which ign) gazebo-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [ruby $(which ign) gazebo-1]: sending signal 'SIGTERM' to process[ruby $(which ign) gazebo-1]
[ERROR] [ruby $(which ign) gazebo-1]: process has died [pid 148, exit code -15, cmd 'ruby $(which ign) gazebo warehouse.sdf -v 4 --gui-config /opt/ros/humble/share/clearpath_gz/config/gui.config --force-version 6'].
```

y2d2, 2023-08-22:
So i see that this has been updated, but has not yet been pushed to the install repos?: https://github.com/clearpathrobotics/clearpath_config/blob/main/clearpath_config/sensors/sensors.py
So I will try to build all clearpath packages form source.

hilary-luo, 2023-08-24:
@y2d2 Thank you for your interest in the clearpath repos. You have jumped aboard early on and astutely have noticed that there are still some issues being worked out. There are patches coming through for a number of packages based on a round of testing that I am just finishing up. See this PR and all of the ones linked: https://github.com/clearpathrobotics/clearpath_common/pull/31

If you are in a rush you can start running the simulation using the indicated branches. If not, these changes are coming soon after a bit more testing, first to main and then will be pushed out to debian.

The documentation website was also recently updated so I would encourage you to revisit that. If you are wanting to try running your Husky using the ROS 2 Humble packages, please see the upgrade instructions here: https://docs.clearpathrobotics.com/docs/ros/installation/upgrading (includes how to create your robot.yaml from your existing envars set on the Husky). However, you do need to be aware that not all sensors are supported yet so please see the list here: https://docs.clearpathrobotics.com/docs/ros/#supported-sensors .

y2d2, 2023-08-28:
Hi @hilary-luo, 
Thanks for the information. Indeed I was expecting the husky I have to be shipped with ROS2 Humble since it seemed like this was now the preferred version, given the warnings for the ROS1 Noetic documentation. However, it arrived with ROS1 and worked, more or less out of the box except for the /scan topic having been remapped to front/scan.
I am now doubting what to do. Should I upgrade to humble I believe the sensors I have are supported in ROS2 per your link. However, I am wondering whether I would not hit a lot of the "children's diseases". 
What would you recommend? 
Thanks 
y2d2

hilary-luo, 2023-08-29:
@y2d2 I can understand the confusion. While we are moving towards ROS 2 Humble as the preferred version, there is some delay between the software being ready and the systems being sold with it. The notice on the website is more to say that we have shifted most development efforts to ROS 2. The decision of whether or not to migrate is really up to your discretion. Noetic is and will continue to be fully supported for the foreseeable future (officially EOL in 2025). That is to say if your system is working in Noetic then you may want to wait until the Clearpath ROS 2 packages are more mature before migrating. However, if there are reasons that you may benefit from working in ROS2, the packages should be stable enough to migrate now and we will do our best to help you address any issues with the packages as they are discovered.

@Cavalletta98 these changes should now be available through the debian packages.

y2d2, 2023-08-29:
@hilary-luo  Thanks A lot for the info! Then I will keep it at ROS1 for now. 

Cheers and thanks!

closed 2023-08-29 by y2d2

Cavalletta98, 2023-09-01:
Ho @hilary-luo still not released the Debian package. I'm still gettin the same errore for Hokuyo laser

tonybaltovski, 2023-09-01:
@Cavalletta98 do you have the Clearpath package server added? It updates faster then the OSRFs.  Here is for `clearpath_common`: https://packages.clearpathrobotics.com/stable/ubuntu/pool/main/r/ros-humble-clearpath-common/  and here is how to setup: http://wiki.ros.org/ClearpathRobotics/Packages

Cavalletta98, 2023-09-04:
@tonybaltovski Done. Now i'm able to see the Hokuyo laser model but not able to see the laser scan in Rviz. I explored the topic and something different from 0 is present. I attach my robot.yaml file. If could help, i attached also the screen of Rviz 
[image]

[robot.zip](https://github.com/clearpathrobotics/clearpath_common/files/12512362/robot.zip)

hilary-luo, 2023-09-05:
@Cavalletta98 Can you please open your own github issue to pursue troubleshooting this? Feel free to assign it to me.

I am able to simulate the robot with your provided robot.yaml file without any problems, including the lidar. In the new issue that you open can you please specify the versions of the packages that you are running and clarify whether you are simulating the robot or running a physical robot? (If you are running a physical robot then please specify the versions of packages that you are running on both the robot and your pc).