clearpathrobotics/clearpath_common #32: Problem to visualize laser scan in simulation

Cavalletta98, 2023-09-06:
Hi, i'm simulating Jackal robot in ROS Humble. I'm able to move it in gazebo via keyboard but not able to see the laser scan in Rviz. The lidar model is Hokuyo and it is visible on the robot. I did `ros2 topic echo /j100_0001/sensors/lidar2d_0/scan` and i saw values equal to 1.0.
Packages version:

- clearpath_common: 0.1.1
- clearpath_simulator: 0.1.1
- clearpath_desktop: 0.1.0
- clearpath_viz: 0.1.0
- clearpath_gz: 0.1.1

I attached my robot.yaml [robot.zip](https://github.com/clearpathrobotics/clearpath_common/files/12534751/robot.zip)
I tag @hilary-luo as mentioned in [30](https://github.com/clearpathrobotics/clearpath_common/issues/30)

hilary-luo, 2023-09-08:
Can you please clarify:
1. are you simulating the robot or running this on a physical robot? I am able to simulate the robot while running the same packages so there is likely something else about your setup.
2. If you are simulating this, are you running the simulation with a GPU?

Cavalletta98, 2023-09-08:
Ho @hilary-luo i'm simulating the robot.
I don't have a dedicate GPU but using the integrating one in the laptop

hilary-luo, 2023-09-13:
Unfortunately that is likely your problem. In order to simulate the lidar properly you will need a dedicated GPU. This warning is on the website but I will make a note that we can make it more prominent / clear.

Cavalletta98, 2023-09-14:
@hilary-luo Is there a way to simulate without dedicated GPU? If i change the type of lidar,Will It works?
Thanks

arthurgomes4, 2023-09-20:
> Hi, i'm simulating Jackal robot in ROS Humble. I'm able to move it in gazebo via keyboard but not able to see the laser scan in Rviz. The lidar model is Hokuyo and it is visible on the robot. I did [...]

The values of `1.0` you  see when  echoing the topic are probably the intensities. If the ranges themselves are 0.0 then your issue may be related to [this](https://github.com/clearpathrobotics/clearpath_simulator/issues/16).

hilary-luo, 2023-09-21:
@Cavalletta98 Please let us know if this solves your issue. @arthurgomes4 Thank you for the contribution!

tonybaltovski, 2023-10-10:
Closing due to inactivity.  Please feel free to re-open if issue persists.

closed 2023-10-10 by tonybaltovski

Cavalletta98, 2023-10-21:
Issue solved. Thanks to @arthurgomes4 . Sorry for beeing late