clearpathrobotics/clearpath_common #40: platform_velocity_controller error

hwlim84, 2023-11-13:
**Describe the the feature you would like**
A clear and concise description of what you want to happen.
OS: ubuntu 22.04 / ROS2 Humble. 
I am trying to run my Husky A200. Tried lots of things but it just ain't working. Please help! Especially on 

[ros2_control_node-2] [ERROR] [1699874992.286789116] [controller_manager]: The 'type' param was not defined for 'platform_velocity_controller'.

user@user-NUC11TNKi5:~$ ros2 launch clearpath_platform platform.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-11-13-20-29-50-933542-user-NUC11TNKi5-9037
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [9040]
[INFO] [ros2_control_node-2]: process started with pid [9042]
[INFO] [spawner-3]: process started with pid [9044]
[INFO] [spawner-4]: process started with pid [9046]
[INFO] [ekf_node-5]: process started with pid [9048]
[INFO] [marker_server-6]: process started with pid [9050]
[INFO] [twist_mux-7]: process started with pid [9052]
[INFO] [joy_linux_node-8]: process started with pid [9054]
[INFO] [teleop_node-9]: process started with pid [9085]
[robot_state_publisher-1] [INFO] [1699874991.383336047] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1699874991.383435643] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1699874991.383444196] [robot_state_publisher]: got segment front_bumper_link
[robot_state_publisher-1] [INFO] [1699874991.383450142] [robot_state_publisher]: got segment front_bumper_mount
[robot_state_publisher-1] [INFO] [1699874991.383455897] [robot_state_publisher]: got segment front_left_wheel
[robot_state_publisher-1] [INFO] [1699874991.383461690] [robot_state_publisher]: got segment front_right_wheel
[robot_state_publisher-1] [INFO] [1699874991.383467337] [robot_state_publisher]: got segment inertial_link
[robot_state_publisher-1] [INFO] [1699874991.383472899] [robot_state_publisher]: got segment rear_bumper_link
[robot_state_publisher-1] [INFO] [1699874991.383478573] [robot_state_publisher]: got segment rear_bumper_mount
[robot_state_publisher-1] [INFO] [1699874991.383483916] [robot_state_publisher]: got segment rear_left_wheel
[robot_state_publisher-1] [INFO] [1699874991.383489323] [robot_state_publisher]: got segment rear_right_wheel
[robot_state_publisher-1] [INFO] [1699874991.383494694] [robot_state_publisher]: got segment top_chassis_link
[robot_state_publisher-1] [INFO] [1699874991.383500113] [robot_state_publisher]: got segment top_plate_front_link
[robot_state_publisher-1] [INFO] [1699874991.383505724] [robot_state_publisher]: got segment top_plate_link
[robot_state_publisher-1] [INFO] [1699874991.383511290] [robot_state_publisher]: got segment top_plate_rear_link
[robot_state_publisher-1] [INFO] [1699874991.383516761] [robot_state_publisher]: got segment user_rail_link
[marker_server-6] [INFO] [1699874991.390950344] [twist_server_node]: [interactive_marker_twist_server] Initialized.
[ros2_control_node-2] [WARN] [1699874991.399810273] [controller_manager]: 'update_rate' parameter not set, using default value.
[ros2_control_node-2] [INFO] [1699874991.400695295] [resource_manager]: Loading hardware 'a200_hardware' 
[ros2_control_node-2] [INFO] [1699874991.402540270] [resource_manager]: Initialize hardware 'a200_hardware' 
[ros2_control_node-2] [INFO] [1699874991.402575410] [A200Hardware]: Name: a200_hardware
[ros2_control_node-2] [INFO] [1699874991.402580295] [A200Hardware]: Number of Joints 4
[teleop_node-9] [INFO] [1699874991.405751000] [TeleopTwistJoy]: Teleop enable button 5.
[teleop_node-9] [INFO] [1699874991.405811442] [TeleopTwistJoy]: Linear axis x on 5 at scale 0.500000.
[teleop_node-9] [INFO] [1699874991.405822349] [TeleopTwistJoy]: Angular axis yaw on 2 at scale 0.500000.
[ros2_control_node-2] [INFO] [1699874991.415801513] [A200Hardware]: Port: /dev/clearpath/prolific
[ros2_control_node-2] [INFO] [1699874991.455166641] [resource_manager]: Successful initialization of hardware 'a200_hardware'
[ros2_control_node-2] [INFO] [1699874991.455222775] [resource_manager]: 'configure' hardware 'a200_hardware' 
[ros2_control_node-2] [INFO] [1699874991.455228383] [resource_manager]: Successful 'configure' of hardware 'a200_hardware'
[ros2_control_node-2] [INFO] [1699874991.455232491] [resource_manager]: 'activate' hardware 'a200_hardware' 
[ros2_control_node-2] [INFO] [1699874991.455236491] [A200Hardware]: Starting ...please wait...
[ros2_control_node-2] [INFO] [1699874991.455239130] [A200Hardware]: System Successfully started!
[ros2_control_node-2] [INFO] [1699874991.455241817] [resource_manager]: Successful 'activate' of hardware 'a200_hardware'
[ros2_control_node-2] [INFO] [1699874991.459622977] [controller_manager]: update rate is 100 Hz
[ros2_control_node-2] [INFO] [1699874991.459764584] [controller_manager]: RT kernel is recommended for better performance
[joy_linux_node-8] [WARN] [1699874991.469273400] [joy_node]: Couldn't open joystick force feedback: Bad file descriptor
[joy_linux_node-8] [INFO] [1699874991.469386044] [joy_node]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ros2_control_node-2] [ERROR] [1699874992.286215948] [controller_manager]: The 'type' param was not defined for 'joint_state_broadcaster'.
[ros2_control_node-2] [ERROR] [1699874992.286789116] [controller_manager]: The 'type' param was not defined for 'platform_velocity_controller'.
[spawner-3] [FATAL] [1699874992.286799913] [spawner_joint_state_broadcaster]: Failed loading controller joint_state_broadcaster
[spawner-4] [FATAL] [1699874992.287343255] [spawner_platform_velocity_controller]: Failed loading controller platform_velocity_controller
[ERROR] [spawner-4]: process has died [pid 9046, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner platform_velocity_controller --ros-args -r __ns:=/'].
[ERROR] [spawner-3]: process has died [pid 9044, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --ros-args -r __ns:=/'].

and snippet of control.yaml

a200_0000:
  controller_manager:
    ros__parameters:
      update_rate: 50
      use_sim_time: False
      joint_state_broadcaster.type: 'joint_state_broadcaster/JointStateBroadcaster'
      platform_velocity_controller.type: 'diff_drive_controller/DiffDriveController'
  platform_velocity_controller:
    ros__parameters:
      use_sim_time: False
      left_wheel_names: ['front_left_wheel_joint', 'rear_left_wheel_joint']
      right_wheel_names: ['front_right_wheel_joint', 'rear_right_wheel_joint']
      wheel_separation: 0.555
      wheels_per_side: 1
      wheel_radius: 0.1651
      wheel_separation_multiplier: 1.875
      left_wheel_radius_multiplier: 1.0
      right_wheel_radius_multiplier: 1.0
      publish_rate: 50.0
      odom_frame_id: 'odom'
      base_frame_id: 'base_link'
      pose_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.01]
      twist_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.01]
      open_loop: False

roni-kreinin, 2023-11-15:
You should not be launching the platform directly like that. Please follow the instructions [here](https://docs.clearpathrobotics.com/docs/ros/installation/robot) for how to set up your robot and have the services launch in the background.

yimlaisum2014, 2023-11-21:
I'm currently experiencing a similar issue and suspect that the controller_manager might not be accessing the /etc/clearpath/platform/config/control.yaml file. In this file, the update rate is set to 50Hz, but when launching control.launch.py, it seems to be operating at 100Hz. Any guidance or suggestions on resolving this would be greatly appreciated.

yimlaisum2014, 2023-11-22:
@hwlim84 
I have successfully addressed this issue by utilizing the absolute path of the config file in my workspace. 
Here are the steps I took to solve the matter:

1. Cloned the 'cleanpath_common,' 'clearpath_config,' and 'clearpath_msg' repositories under my workspace.
2. Modified the configuration paths in 'control.launch.py' and 'teleop_joy.launch.py' accordingly.
3. Finally 'colcon build --symlink-install' 

Hope these steps prove helpful. 
_______
By the way, @roni-kreinin  I would like to inquire whether this modification would have any bad effects on subsequent operations?

roni-kreinin, 2023-11-22:
@yimlaisum2014 It would be better to figure out why the controller manager is not reading the `/etc/clearpath/platform/config/control.yaml` file. The intention is for you to be able to modify control parameters in your `robot.yaml` file, rather than having to modify the launch file from source.

When you say 

> when launching control.launch.py

do you mean that you are manually calling `ros2 launch clearpath_control control.launch.py`?

closed 2023-12-20 by tonybaltovski