clearpathrobotics/clearpath_common #56: Error in automatic URDF generation from YAML description

fbottarel, 2024-01-09:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble Hawksbill
 - Built from source or installed: installed 
 - Package version: 0.2.2-1jammy.20240104.205652 amd64
 - Real hardware or simulation: real

 **Expected behaviour**
The `clearpath-robot-generate` executable script generates a functioning `robot.urdf.xacro`. In my case, from this `robot.yaml` config file 

<details>
  <summary>my robot.yaml</summary>

```yaml
serial_number: a200-0000
version: 0
system:
  username: administrator
  hosts:
    self: lattepanda3delta
    platform:
      lattepanda3delta: 192.168.131.1
    onboard: {}
    remote: {}
  ros2:
    namespace: a200_0000
    domain_id: 20
    rmw_implementation: rmw_fastrtps_cpp
    workspaces: []
platform:
  controller: logitech
  battery:
    model: ES20_12C
    configuration: S2P1
  attachments: []
links:
  box: []
  cylinder: []
  frame: []
  mesh: []
  sphere: []
mounts:
  bracket: []
  fath_pivot: []
  riser: []
  disk: []
  post: []
sensors:
  camera: []
  gps: []
  imu: []
  lidar2d:
  - model: hokuyo_ust
    urdf_enabled: true
    launch_enabled: true
    parent: base_link
    xyz: [0.0, 0.0, 0.2]
    rpy: [0.0, 0.0, 0.0]
    ros_parameters:
      urg_node:
        laser_frame_id: lidar2d_0_laser
        ip_address: 192.168.131.20
        ip_port: 10940
        angle_min: -2.356
        angle_max: 2.356
  lidar3d: []
```

</details>

I'm expecting a `robot.urdf.xacro` like 

<details>
<summary>expected robot.urdf.xacro</summary>

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="a200-0000">

  <!-- Common -->
  <xacro:include filename="$(find clearpath_platform_description)/urdf/common.urdf.xacro"/>

  <!-- Platform -->
  <xacro:include filename="$(find clearpath_platform_description)/urdf/a200/a200.urdf.xacro"/>
  <xacro:a200 wheel="default"/>

  <!-- Attachments -->

  <!-- Links -->

  <!-- Mounts -->

  <!-- Sensors -->

  <!-- lidar2d_0 -->
  <xacro:include filename="$(find clearpath_sensors_description)/urdf/hokuyo_ust.urdf.xacro"/>
  <xacro:hokuyo_ust name="lidar2d_0" parent_link="base_link" ang_res="0.5" min_ang="-2.356" max_ang="2.356" min_range="0.05" max_range="25.0" update_rate="50">
    <origin xyz="0.0 0.0 0.2" rpy="0.0 0.0 0.0"/>
  </xacro:hokuyo_ust>

  <!-- Extras -->

</robot>

```

</details>

 **Actual behaviour**
The `robot.urdf.xacro` file that is being produced tries to include a `.urdf.xacro` file that is missing even when the field `urdf` in the YAML is empty or missing. This causes a failure in URDF parsing and therefore the platform launch files cannot be run.

<details>
<spoiler>actual robot.urdf.xacro</spoiler>

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="a200-0000">

  <!-- Common -->
  <xacro:include filename="$(find clearpath_platform_description)/urdf/common.urdf.xacro"/>

  <!-- Platform -->
  <xacro:include filename="$(find clearpath_platform_description)/urdf/a200/a200.urdf.xacro"/>
  <xacro:a200 wheel="default"/>

  <!-- Attachments -->

  <!-- Links -->

  <!-- Mounts -->

  <!-- Sensors -->

  <!-- lidar2d_0 -->
  <xacro:include filename="$(find clearpath_sensors_description)/urdf/hokuyo_ust.urdf.xacro"/>
  <xacro:hokuyo_ust name="lidar2d_0" parent_link="base_link" ang_res="0.5" min_ang="-2.356" max_ang="2.356" min_range="0.05" max_range="25.0" update_rate="50">
    <origin xyz="0.0 0.0 0.2" rpy="0.0 0.0 0.0"/>
  </xacro:hokuyo_ust>

  <!-- Extras -->

  <xacro:include filename=".urdf.xacro"/>

</robot>

```

</details>

**To Reproduce**
Provide the steps to reproduce:
1. Copy my `robot.yaml` in the `/etc/clearpath` directory after the recommended PC install procedure
2. Run the `clearpath-robot-generate` script to generate the URDF and launch files
3. Examine the resulting `robot.urdf.xacro` under `/etc/clearpath`.

**Other notes**
Trying to examine and reproduce the issue, it looks like this `if` statement always executes even if there is no `extra` or `urdf` field in the `robot.yaml`

https://github.com/clearpathrobotics/clearpath_common/blob/10fa253d51fea9c9879f72a6e013961def68186d/clearpath_generator_common/clearpath_generator_common/description/generator.py#L202C1-L207

hilary-luo, 2024-01-09:
Which version of clearpath_config are you running? This sounds like the bug that was patched yesterday, try updating your packages and see if it persists.

fbottarel, 2024-01-10:
I just updated `ros-humble-clearpath-config` to `0.2.2-1jammy.20240108.190424` and seems like the latest merge solves the issue. Thanks @hilary-luo!

closed 2024-01-10 by fbottarel