clearpathrobotics/clearpath_common #83: Error in generating the launch files from the "robot.yaml" because of the generate_manipulators() method

aalmrad, 2024-08-23:
**Please provide the following information:**
 - OS: Ubuntu 22.04.4 LTS
 - ROS 2 Distro: Humble
 - Built from source
 - Package version: commit 8974d1e8845c5c56b91a465a5f3689e5001fcb11
 - Simulation

 **Expected behaviour**
Start the simulation launch file that will use the generators to create the launch files and the robot description using the parameters specified in the "robot.yaml" file. Then the simulation environment with the spawned robot will start.

 **Actual behaviour**
The launch files are not generated because of the NotImplemented generate_manipulators() method ERROR.

**To Reproduce**
Provide the steps to reproduce:
1. Follow the instructions here to build the workspace: https://docs.clearpathrobotics.com/docs/ros/tutorials/simulator/install
2. Create the ~/clearpath directory and add the following "robot.yaml" file:
```
serial_number: a200-0000
version: 0
system:
  username: administrator
  hosts:
    - hostname: cpr-a200-0000
      ip: 192.168.131.1
    - hostname: offboard-computer
      ip: 192.168.131.5
  ros2:
    namespace: a200_0000
    domain_id: 0
    middleware:
      implementation: rmw_fastrtps_cpp
      discovery: simple
    workspaces:
      - /home/aalmrad/clearpath_ws/install/setup.bash
platform:
  attachments:
    - name: front_bumper
      type: a200.bumper
      model: default
      parent: front_bumper_mount
      xyz: [0.0, 0.0, 0.0]
      rpy: [0.0, 0.0, 0.0]
      enabled: true
      extension: 0.0
  
  battery:
    model: ES20_12C
    configuration: S2P1
  controller: ps4 # or logitech

links:
  box:
  - name: user_bay_cover
    parent: default_mount
    xyz: [0.0, 0.0, 0.0075]
    rpy: [0.0, 0.0, 0.0]
    size: [0.4, 0.4, 0.002]
#mounts:
#  bracket:
#  - parent: top_plate_mount_d1
#    xyz: [0.0, 0.0, 0.0]
#    rpy: [0.0, 0.0, 0.0]
#    model: horizontal
#  fath_pivot:
#  - parent: sensor_arch_mount
#   xyz: [0.0, 0.0, -0.021]
#   rpy: [3.1415, 0.0, 0.0]
#   angle: 0.0
#sensors:
#  camera:
# - model: intel_realsense
#    urdf_enabled: true
#    launch_enabled: true
#    parent: fath_pivot_0_mount
#    xyz: [0.0, 0.0, 0.0]
#    rpy: [0.0, 0.0, 0.0]
#    ros_parameters:
#      intel_realsense:
#        camera_name: camera_0
#        device_type: d435
#        serial_no: '0'
#        enable_color: true
#        rgb_camera.profile: 640,480,30
#        enable_depth: true
#        depth_module.profile: 640,480,30
#        pointcloud.enable: true
#  lidar2d:
#  - model: hokuyo_ust
#    urdf_enabled: true
#    launch_enabled: true
#    parent: bracket_0_mount
#    xyz: [0.0, 0.0, 0.0]
#    rpy: [0.0, 0.0, 0.0]
#    ros_parameters:
#      urg_node:
#        laser_frame_id: lidar2d_0_laser
#        ip_address: 192.168.131.20
#        ip_port: 10940
#        angle_min: -3.141592653589793
#        angle_max: 3.141592653589793
#  lidar3d:
#  - model: velodyne_lidar
#    urdf_enabled: true
#    launch_enabled: true
#    parent: sensor_arch_mount
#    xyz: [0.0, 0.0, 0.0]
#    rpy: [0.0, 0.0, 0.0]
#    ros_parameters:
#      velodyne_driver_node:
#        frame_id: lidar3d_0_laser
#        device_ip: 192.168.131.25
#        port: 2368
#        model: VLP16
#      velodyne_transform_node:
#        model: VLP16
#        fixed_frame: lidar3d_0_laser
#        target_frame: lidar3d_0_laser
```
3. In a new terminal window, use the following command:
```
ros2 launch clearpath_gz simulation.launch.py
```

tonybaltovski, 2024-09-12:
@aalmrad this was fixed here: https://github.com/clearpathrobotics/clearpath_robot/commit/9dd6a3eb35a66610331f7497f89c1e3060ba7c41

closed 2024-09-12 by tonybaltovski

aalmrad, 2024-09-13:
@tonybaltovski thank you, i will test it soon.