clearpathrobotics/clearpath_common #85: Unable to find clearpath_platform_msg package

iamkashish, 2024-09-16:
I'm using humble branch of this repository on Ubuntu 22.04 and with ROS 2 humble.

While doing `colcon build`:

--- stderr: clearpath_platform                                  
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findclearpath_platform_msgs.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "clearpath_platform_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "clearpath_platform_msgs" with any of the following names:

    clearpath_platform_msgsConfig.cmake
    clearpath_platform_msgs-config.cmake

  Add the installation prefix of "clearpath_platform_msgs" to
  CMAKE_PREFIX_PATH or set "clearpath_platform_msgs_DIR" to a directory
  containing one of the above files.  If "clearpath_platform_msgs" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< clearpath_platform [1.36s, exited with code 1]

++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Then, I checked there is no folder named clearpath_platform_msgs. Also I tried making using the humble_msgs from https://github.com/husky/husky.git but it ends up with other issues.

tonybaltovski, 2024-09-17:
@iamkashish did you use `rosdep` to install the dependencies prior to bulding? While in your workspace, run: `rosdep install --from-paths src --ignore-src --rosdistro=$ROS_DISTRO -y`.  Else, can you install it manually using: `sudo apt install ros-humble-clearpath-platform-msgs`?

iamkashish, 2024-09-17:
Yes I did. I got the following:

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
clearpath_manipulators: Cannot locate rosdep definition for [position_controllers]
clearpath_manipulators_description: Cannot locate rosdep definition for [robotiq_description]
clearpath_control: Cannot locate rosdep definition for [joy_linux]
clearpath_sensors_description: Cannot locate rosdep definition for [velodyne_description]
clearpath_generator_common: Cannot locate rosdep definition for [moveit_setup_srdf_plugins]
clearpath_platform: Cannot locate rosdep definition for [puma_motor_msgs]
`

tonybaltovski, 2024-09-18:
Is your rosdep working at all?  It looks like you can't really find anything, not just Clearpath specific packages.

iamkashish, 2024-09-18:
I don't know what happened last time because just now I tried again and it installed ros-humble-clearpath-platform-msgs

But now while doing `colcon build`:

`/main_ws/src/husky_ws/src/clearpath_platform/src/lighting/lighting.cpp: In member function ‘void clearpath_lighting::Lighting::updateState()’:
/main_ws/src/husky_ws/src/clearpath_platform/src/lighting/lighting.cpp:416:24: error: ‘using StopStatus = struct clearpath_platform_msgs::msg::StopStatus_<std::allocator<void> >’ {aka ‘struct clearpath_platform_msgs::msg::StopStatus_<std::allocator<void> >’} has no member named ‘needs_reset’
  416 |   if (stop_status_msg_.needs_reset)
      |                        ^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/clearpath_platform_lighting.dir/build.make:76: CMakeFiles/clearpath_platform_lighting.dir/src/lighting/lighting.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/clearpath_platform_lighting.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< clearpath_platform [13.3s, exited with code 2]

Summary: 19 packages finished [14.3s]
  1 package failed: clearpath_platform
  1 package had stderr output: clearpath_platform
  2 packages not processed
`

tonybaltovski, 2024-09-19:
We are working a release at the moment.  You are going to needed to install the messages from source as well because there was a change to them.

iamkashish, 2024-09-24:
@tonybaltovski I know it's not a right question to ask but I have to plan my project accordingly, so can I know what is the 
probable expected date of the release?
Thank you very much for your support!

tonybaltovski, 2024-09-24:
@iamkashish was just released.

tonybaltovski, 2024-10-02:
Have you updated your packages?  Messages should be on `0.3`  Try cleaning your workspace.

iamkashish, 2024-10-02:
Yes, I updated the packages. I have the messages but the error is about 'needs_reset' (e.g. has no member named needs_reset)
I install the dependencies again. 
My dependencies.repos file:  
`repositories:
  clearpath_config:
    type: git
    url: https://github.com/clearpathrobotics/clearpath_config.git
    version: main
  clearpath_msgs:
    type: git
    url: https://github.com/clearpathrobotics/clearpath_msgs.git
    version: humble
` 
Cleaned the workspace. Then rebuild again. I get below errors

Error:
`--- stderr: clearpath_platform                                 
/main_ws/src/clearpath_common-humble/clearpath_platform/src/lighting/lighting.cpp: In member function ‘void clearpath_lighting::Lighting::updateState()’:
/main_ws/src/clearpath_common-humble/clearpath_platform/src/lighting/lighting.cpp:416:24: error: ‘using StopStatus = struct clearpath_platform_msgs::msg::StopStatus_<std::allocator<void> >’ {aka ‘struct clearpath_platform_msgs::msg::StopStatus_<std::allocator<void> >’} has no member named ‘needs_reset’
  416 |   if (stop_status_msg_.needs_reset)
      |                        ^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/clearpath_platform_lighting.dir/build.make:76: CMakeFiles/clearpath_platform_lighting.dir/src/lighting/lighting.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/clearpath_platform_lighting.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< clearpath_platform [44.8s, exited with code 2]

Summary: 12 packages finished [48.3s]
  1 package failed: clearpath_platform
  1 package had stderr output: clearpath_platform
  2 packages not processed
`

tonybaltovski, 2024-10-02:
Can you post the full build log please?

iamkashish, 2024-10-04:
```
Starting >>> clearpath_manipulators_description
Starting >>> clearpath_platform_description
Starting >>> clearpath_control
Starting >>> clearpath_mounts_description
Starting >>> clearpath_sensors_description
Starting >>> rexbot_flexbe_behaviors
Starting >>> rexbot_flexbe_states
Starting >>> btcpp_sample
Finished <<< clearpath_mounts_description [1.12s]
Finished <<< clearpath_manipulators_description [1.17s]
[2.699s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'btcpp_sample' because it has no 'install' target
Finished <<< clearpath_control [1.16s]
Starting >>> clearpath_customization
Starting >>> clearpath_manipulators   
Finished <<< clearpath_sensors_description [1.19s]
Finished <<< clearpath_platform_description [1.23s]
Starting >>> clearpath_description
Starting >>> clearpath_platform
Finished <<< btcpp_sample [1.24s]                                              
Finished <<< rexbot_flexbe_behaviors [1.41s]                                   
Finished <<< clearpath_customization [0.34s]                        
Finished <<< clearpath_description [0.31s]
Finished <<< clearpath_manipulators [0.40s]
Finished <<< rexbot_flexbe_states [2.62s]                            
Starting >>> rexbot_behaviors
Finished <<< rexbot_behaviors [0.18s]                                        
--- stderr: clearpath_platform                                 
/main_ws/src/clearpath_common-humble/clearpath_platform/src/lighting/lighting.cpp: In member function ‘void clearpath_lighting::Lighting::updateState()’:
/main_ws/src/clearpath_common-humble/clearpath_platform/src/lighting/lighting.cpp:416:24: error: ‘using StopStatus = struct clearpath_platform_msgs::msg::StopStatus_<std::allocator<void> >’ {aka ‘struct clearpath_platform_msgs::msg::StopStatus_<std::allocator<void> >’} has no member named ‘needs_reset’
  416 |   if (stop_status_msg_.needs_reset)
      |                        ^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/clearpath_platform_lighting.dir/build.make:76: CMakeFiles/clearpath_platform_lighting.dir/src/lighting/lighting.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/clearpath_platform_lighting.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< clearpath_platform [23.6s, exited with code 2]

Summary: 12 packages finished [25.9s]
  1 package failed: clearpath_platform
  1 package had stderr output: clearpath_platform
  2 packages not processed
```

This is the full build log.

VictorVigara, 2024-10-07:
I had the same issue and I have been able to solve it by removing the messages package installed by rosdep (version 0.2) with this command: 
`sudo apt-get remove --purge ros-humble-clearpath-msgs`

And then cloning the repository which contains the version 0.3: 
`git clone https://github.com/clearpathrobotics/clearpath_msgs.git --branch humble`

tonybaltovski, 2024-10-09:
@iamkashish I was looking for underlay workspaces but that wasn't captured in your output.  The error is due to building against 0.2 version of messages but the CI builds this from source and was successful.  Using the OSRFs main package server, it fails with the same way but the testing passes.  You can switch to testing and not build from source.  Also, you can add the Clearpath package server which usually has more update-to-date packages.

closed 2024-11-28 by tonybaltovski