clearpathrobotics/clearpath_common #106: Avoid prefixing tf and tf_static when using namespaces

taDachs, 2024-11-07:
**Describe the the feature you would like**
It would be preferable to not prefix the tf and tf_static topics when using the `namespace` argument.

**Other notes**
tf2_ros by default subscribes to the global tf and tf_static topics. Prefixing those forces us to remap tf in every node/launch file that we add to the stack, or relaying the /namespace/tf to /tf using a relay node.

If I understand correctly, the reason why tf gets prefixed is the use of the `PushRosNamespace` launch action, which also prefixes globally namespaced topics.

SOHAILMBS, 2024-11-08:
I believe they are doing this for multi-robot systems on the same domain id.

taDachs, 2024-11-08:
I get your point, however, if you have a look at https://github.com/clearpathrobotics/clearpath_common/blob/2f04d13cd607b0249032f3812101e5855592bf5c/clearpath_control/launch/control.launch.py#L50 

`tf` and `tf_static` are explicity remapped from global topics, as it should be, to non-global topics just so they can be namespaced.

We do multirobot teams, and we are using [the domain bridge](https://github.com/ros2/domain_bridge) for bridging topics between domains. I don't know why you would want prefixed tf topics instead of just prefixing your tf frames

hilary-luo, 2024-11-13:
@taDachs The prefixing of the tf and tf_static topics is an intentional decision and yes has been explicitly done. I would encourage you to share your opinions on why it should not be done so that it is recorded here, but there are also many reasons why one may not want the tf trees all connected. You have taken a certain approach in using domain bridging to build your fleet and I congratulate you on finding something that works for you, but that is not suitable for everyone and has its own limitations. 

This is a much bigger discussion but I can give you a couple of quick reasons why we format it this way. First off, as was said, it provides the ability to scale up to a large fleet without using domain bridging. Additionally, this lines up with the standards used by Nav2, iRobot and our Turtlebot 4 software. This allows our system to work more seamlessly with the Nav2 packages in particular. There are several trails of different tickets discussing these issues across multiple repos and it is intermixed with discussion about the deprecation of tf_prefix. I would encourage you to explore these discussions further and understand more aspects and implications of this decision for different groups.

Although we will not be acting on your suggestion, I do want to thank you for sharing it here.

closed 2024-11-13 by hilary-luo

taDachs, 2024-11-23:
@hilary-luo thanks for the response and the explanation. 

I get that our approach for multi robot setups is unique to us, but remapping tf that way through the whole hardware stack of the robot results in a lot of headaches where one doesn't use tf that way. 

One example of that is our stack: We would have to adapt all of our launch files to also remap tf that way. Our setup includes robots of different types and manufactures, so we would have to go through a lot of work just to get our stack compatible with husky, or keep an entire separate package with launch files specifically for husky that mirror the launch files used by the other robots except for that one detail. (The other robots use a globally namespaced tf/tf_static).

The last few days I tried to use a different approach of just forking the `clearpath_common` repos and removing the namespaces for tf, but then I run into problems where some of the launch files are not open source, meaning I have to either add relays for that topic or take the launch files from the global install space, move them to a new package, and then remove the namespaces. This not only feels incredibly sketchy, but also means more work in the future for keeping those launch files up-to-date. Also, sometimes I can't even find where the namespaces are located, for example when trying to work with `clearpath_simulator`.

 I get that you have your reasons for doing it that way, but not having the option to do it another way (that isn't uncommon) makes the setup very frustrating. 

Would it maybe be possible to make it conditional via a launch file argument?