clearpathrobotics/clearpath_common #121: platform_velocity_controller publishes tf as TRANSIENT_LOCAL

taDachs, 2024-11-28:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Humble
 - Built from source or installed: tried both
 - Package version: 1.0.0-1jammy.20241126.044506 (apt), 2ac65c79f489732a95d1951a57a31d7461bc21d1 (source)
 - Real hardware or simulation: hardware, tried with mock hardware as well (ros2_control mock hardware, not simulation)

 **Expected behaviour**
Normal tf (not tf static) should be published VOLATILE

 **Actual behaviour**
`platform_velocity_controller` publishes tf as TRANSIENT_LOCAL

**To Reproduce**
Start the platform_velocity_controller and check the QoS for tf

**Other notes**
Output of `ros2 topic info -v /husky/tf`

```
Node name: platform_velocity_controller
Node namespace: /husky
Topic type: tf2_msgs/msg/TFMessage
Endpoint type: PUBLISHER
GID: 01.0f.57.0b.0d.89.5f.73.00.00.00.00.00.00.4f.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
  ```
  
 `/tf` (or in this case `/husky/tf`) should always be published as VOLATILE, as TRANSIENT_LOCAL latches the topic and old tf messages have no value. It also can lead to error messages because of dropped messages. It is not a critical issue or anything but should still be mentioned

Mechazo11, 2024-12-05:
Hi @taDachs I was wondering does this error lead to `TF_OLD_DATA` warning message?

taDachs, 2024-12-05:
@Mechazo11 hey, yes, this can happen. `TF_OLD_DATA` gets thrown when data is added to your tf_buffer that is older than the cache limit (see [here](http://wiki.ros.org/tf/Errors%20explained)). By latching the topic old messages might get added to your tf buffer, resulting in that error.

tonybaltovski, 2024-12-09:
@taDachs this looks deeper than this repo.  It looks to be from here: https://github.com/ros-controls/ros2_controllers/blob/master/diff_drive_controller/src/diff_drive_controller.cpp#L412  Do you want to open an issue there?

closed 2025-01-14 by tonybaltovski