clearpathrobotics/clearpath_common #124: Seeking clarificaiton of the last two lines after top-level Software License Agreement. Do I need formal permission to publish a derived work on clearpath_simulator?

Mechazo11, 2024-12-05:
Hello Clearpath Robotics,

In a number of `.py` files (like ```/launch/writer.py```, ```/description/generate_description.py```) after the ```Software License Agreement``` docstring,  two lines are present as shown below

```python
# Redistribution and use in source and binary forms, with or without
# modification, is not permitted without the express permission
# of Clearpath Robotics.
``` 
As far as I understand, BSD-3 license allows redistribution as long as the conditions for redistribution are met and the "AS IS" notice.

```python
* Redistributions of source code must retain the above copyright notice,
  this list of conditions and the following disclaimer.
* Redistributions in binary form must reproduce the above copyright notice,
  this list of conditions and the following disclaimer in the documentation
  and/or other materials provided with the distribution.
* Neither the name of Clearpath Robotics nor the names of its contributors
  may be used to endorse or promote products derived from this software
  without specific prior written permission.
```

At the time of writing this message, ```clearpath_common``` is released with BSD-3 license (referring to Version 1.0): https://github.com/clearpathrobotics/clearpath_common/blob/humble/LICENSE

I have been working with ```clearpath_simulator``` for the last two months and have made substantial changes to make it compatible with a newer version of Gazebo. My copy of ```clearpath_common``` thus underwent a number of changes to achieve this compatibility.

My questions thus are follows
a. Are those two lines contradicting the BSD-3 license?
b. If the above two lines are valid even in context of being BSD-3 license, how can I obtain a formal permission from Clearpath Robotics to publish my work?

tonybaltovski, 2024-12-05:
1. They should all be this: https://github.com/clearpathrobotics/headers/blob/master/bsd/py so those lines are contradicting the BSD-3 license.
2. No needs, feel free to submit PRs.

Mechazo11, 2024-12-05:
Hi @tonybaltovski 

Thank you very much for your prompt response. 

1. I shall remove those lines from my fork of ```clearpath_common``` from the files. The rest conforms to the template you provided.

2. I am a bit hesitant about releasing PRs just yet since I modified a number of packages to make them compatible with the newer Gazebo. I plan on releasing the project separately today and hopefully you and your team may assess whether it would be worthwhile to integrate into the official version.

With best,
@Mechazo11

closed 2024-12-05 by Mechazo11