clearpathrobotics/clearpath_common #247: Discovery Server not working in ROS2 Jazzy

pragnaISAUoY, 2025-08-11:
**Please provide the following information:**
 - OS: Ubuntu 24.04
 - ROS 2 Distro: Jazzy
 - Built from source or installed: Built from Source 
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_DISTRO-clearpath-common`, if from source, give commit hash)
 - Real hardware or simulation: Real Hardware

 **Expected behaviour**
 Discovery Server mode will work and ros2 topics will be available in offboard computer 

 **Actual behaviour**
 All ros2 communication stops and COMM light is RED when Discovery Server mode is made active. Even joystick controls is not communication to robot. ros2 topic are not returned with ros2 topic list

**To Reproduce**
un : ls 
You will see the robot.yaml file 
clearpath_computer_installer.sh  clearpath_ws  profile_dds.xml  robot.yaml
Run: vi robot.yaml,
Make necessary changes, press ESC AND wq!
This will save the file
Run: cp robot.yaml /etc/clearpath/robot.yaml
To copy this file in /etc/clearpath
Run: sudo systemctl daemon-reload && sudo systemctl start clearpath-robot.service
To re-start services 
This script used[ robot_upstart](https://github.com/clearpathrobotics/robot_upstart/tree/foxy-devel) to install the systemd services that will launch platform and sensor launch files as background processes.

See the result: COMM light turns RED and no ros2 communication happens, Even joystick controls is not communication to robot. ros2 topic are not returned with ros2 topic list

**Other notes**
Add anything else you think is important.

pragnaISAUoY, 2025-08-12:
When in Discovery : Simple mode, the teleoperation with keyboard and joystick works, with COMM light being green,
When in Discovery: server, the COMM turns RED, the teleoperation with keyboard and joystick stops, and ros2 topic list doesnt show topics

tonybaltovski, 2025-08-25:
Closing due to inactivity.  Please feel free to re-open if you would like to continue the troubleshooting process.