clearpathrobotics/clearpath_common #250: Controller "Type param error"

shrandy03, 2025-08-19:
**Please provide the following information:**
 - OS: Ubuntu 24.04
 - ROS 2 Distro: Jazzy
 - Built from source or installed: Installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_DISTRO-clearpath-common`, if from source, give commit hash)
 - Real hardware or simulation: Simulation
[image]

I am trying to run ridgeback robot. but I am constantly receiving the bellow error
Command : ros2 launch bringup_package simulation.launch.py

[image]

detailed log output:
[INFO] [launch]: All log files can be found below /home/sandy/.ros/log/2025-08-19-08-53-26-052809-4bc820d0ba5d-13982
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [13999]
[INFO] [parameter_bridge-2]: process started with pid [14000]
[INFO] [robot_state_publisher-3]: process started with pid [14002]
[INFO] [create-4]: process started with pid [14003]
[robot_state_publisher-3] [INFO] [1755593606.935582724] [r100_0000.robot_state_publisher]: Robot initialized
[create-4] [INFO] [1755593606.946304137] [r100_0000.ros_gz_sim]: Requesting list of world names.
[parameter_bridge-2] [INFO] [1755593607.000635682] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (gz.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[gazebo-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-sandy'
[gazebo-1] [Msg] Gazebo Sim GUI    v8.9.0
[gazebo-1] [Dbg] [Gui.cc:275] Waiting for subscribers to [/gazebo/starting_world]...
[gazebo-1] [Dbg] [Application.cc:96] Initializing application.
[gazebo-1] [Dbg] [Application.cc:170] Qt using OpenGL graphics interface
[gazebo-1] [GUI] [Dbg] [Application.cc:657] Create main window
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Wood
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Wrn] [SdfEntityCreator.cc:933] Gazebo does not support Ogre material scripts. See https://gazebosim.org/api/sim/8/migrationsdf.html#:~:text=Materials for details.
[gazebo-1] [Wrn] [SdfEntityCreator.cc:949] Using an internal gazebo.material to parse Gazebo/Grey
[gazebo-1] [Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[gazebo-1] [Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[gazebo-1] [Msg] Received world [warehouse.sdf] from the GUI.
[gazebo-1] [Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[gazebo-1] [Msg] Gazebo Sim Server v8.9.0
[gazebo-1] [Msg] Loading SDF world file[/workspace/install/samxl_mobile_manipulator_bringup/share/samxl_mobile_manipulator_bringup/worlds/warehouse.sdf].
[gazebo-1] [Msg] Serving entity system service on [/entity/system/add]
[gazebo-1] [Dbg] [Physics.cc:883] Loaded [gz::physics::dartsim::Plugin] from library [/opt/ros/jazzy/opt/gz_physics_vendor/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[gazebo-1] [Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::Physics] for entity [1]
[gazebo-1] [Msg] Create service on [/world/warehouse/create]
[gazebo-1] [Msg] Remove service on [/world/warehouse/remove]
[gazebo-1] [Msg] Pose service on [/world/warehouse/set_pose]
[gazebo-1] [Msg] Pose service on [/world/warehouse/set_pose_vector]
[gazebo-1] [Msg] Light configuration service on [/world/warehouse/light_config]
[gazebo-1] [Msg] Physics service on [/world/warehouse/set_physics]
[gazebo-1] [Msg] SphericalCoordinates service on [/world/warehouse/set_spherical_coordinates]
[gazebo-1] [Msg] Enable collision service on [/world/warehouse/enable_collision]
[gazebo-1] [Msg] Disable collision service on [/world/warehouse/disable_collision]
[gazebo-1] [Msg] Material service on [/world/warehouse/visual_config]
[gazebo-1] [Msg] Material service on [/world/warehouse/wheel_slip]
[gazebo-1] [Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[gazebo-1] [Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[gazebo-1] [Dbg] [Sensors.cc:697] Configuring Sensors system
[gazebo-1] [Dbg] [Sensors.cc:557] SensorsPrivate::Run
[gazebo-1] [Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::Sensors] for entity [1]
[gazebo-1] [Dbg] [Sensors.cc:532] SensorsPrivate::RenderThread started
[gazebo-1] [Dbg] [Sensors.cc:337] Waiting for init
[gazebo-1] [Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::Imu] for entity [1]
[gazebo-1] [Dbg] [SystemManager.cc:80] Loaded system [gz::sim::systems::NavSat] for entity [1]
[gazebo-1] [Msg] Loaded level [default]
[gazebo-1] [Msg] Serving world controls on [/world/warehouse/control], [/world/warehouse/control/state] and [/world/warehouse/playback/control]
[... 170 lines cut ...]
[gazebo-1] [Dbg] [SDFFeatures.cc:332] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[gazebo-1] [Dbg] [SDFFeatures.cc:864] The geometry element of collision [base_link_fixed_joint_lump__hams_base_link_collision_4] couldn't be created
[gazebo-1] [Dbg] [SDFFeatures.cc:332] Mesh construction from an SDF has not been implemented yet for dartsim. Use AttachMeshShapeFeature to use mesh shapes.
[gazebo-1] [Dbg] [SDFFeatures.cc:864] The geometry element of collision [base_link_fixed_joint_lump__top_link_collision_15] couldn't be created
[gazebo-1] [Dbg] [SimulationFeatures.cc:90] Simulation timestep set to: 0.003
[gazebo-1] [Dbg] [Sensors.cc:953] Initialization needed
[gazebo-1] [Dbg] [Sensors.cc:349] Initializing render context
[gazebo-1] [Msg] Loading plugin [gz-rendering-ogre2]
[INFO] [spawner-5]: process started with pid [14295]
[INFO] [spawner-6]: process started with pid [14296]
[gazebo-1] [WARN] [1755593611.904686923] [r100_0000.gz_ros_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.003 s).
[gazebo-1] libEGL warning: failed to open /dev/dri/renderD128: Permission denied
[gazebo-1] 
[gazebo-1] libEGL warning: failed to open /dev/dri/renderD128: Permission denied
[gazebo-1] 
[gazebo-1] [Wrn] [Component.hh:144] Trying to serialize component with data type [St6vectorIdSaIdEE], which doesn't have `operator<<`. Component will not be serialized.
[gazebo-1] [ERROR] [1755593612.975307890] [r100_0000.controller_manager]: The 'type' param was not defined for 'joint_state_broadcaster'.
[spawner-5] [FATAL] [1755593613.019980731] [r100_0000.spawner_joint_state_broadcaster]: Failed loading controller joint_state_broadcaster
[ERROR] [spawner-5]: process has died [pid 14295, exit code 1, cmd '/opt/ros/jazzy/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /r100_0000/controller_manager --controller-manager-timeout 120 --ros-args -r __node:=spawner_joint_state_broadcaster -r __ns:=/r100_0000'].
[gazebo-1] [ERROR] [1755593613.353654221] [r100_0000.controller_manager]: The 'type' param was not defined for 'platform_velocity_controller'.
[spawner-6] [FATAL] [1755593613.419326636] [r100_0000.spawner_platform_velocity_controller]: Failed loading controller platform_velocity_controller

This is in my control.yaml file

[image]

tonybaltovski, 2025-08-22:
Are you running `ros2 launch clearpath_gz simulation.launch.py setup_path:=$HOME/setup/path/`?  I am not sure where ` bringup_package`  in `ros2 launch bringup_package simulation.launch.py` comes from.

shrandy03, 2025-08-22:
Oh! yeah I am running this command ros2 launch clearpath_gz simulation.launch.py setup_path:=$HOME/setup/path/. It was a mistake on my part.

shrandy03, 2025-09-01:
Hi @tonybaltovski,  

[output.log](https://github.com/user-attachments/files/22074867/output.log)

[robot.yaml](https://github.com/user-attachments/files/22074874/robot.yaml)

[control.yaml](https://github.com/user-attachments/files/22074880/control.yaml)

Unfortunately, I am still encountering the same errors related to joint_state_broadcaster and platform_velocity_controller.

For clarity, I have attached the following files for your reference:

- Full log output (output.log)
- Controllers configuration YAML
- Robot YAML

I have followed the setup instructions from both the Clearpath website and the clearpath_simulator repository. While the simulation launches successfully, the robot does not move. To troubleshoot, I also added the Gazebo ros2 control plugins in the robot description, but the issue persists.

Please let me know if I should try additional steps or provide further details.