clearpathrobotics/clearpath_common #252: Missing package ewellix_description breaks builds

vtomnet, 2025-08-27:
**Please provide the following information:**
 - OS: Ubuntu 24.04
 - ROS 2 Distro: Jazzy
 - Built from source or installed: installed
 - Package version: fb5b2a1
 - Real hardware or simulation: docker

 **Expected behaviour**

All dependencies of clearpath_common should install; ewellix_description does not. Relevant PR opened and merged two days ago: https://github.com/clearpathrobotics/clearpath_common/pull/251.

 **Actual behaviour**

```
executing command [apt-get install -y ros-jazzy-ewellix-description]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-jazzy-ewellix-description
```

Indeed, we can see in the [PPA package list](http://packages.ros.org/ros2/ubuntu/lists/ros-jazzy-noble-amd64_noble_main_amd64_Packages) that the `ewellix-description` package is not present. This same error was reported by the CI test in the PR: https://github.com/clearpathrobotics/clearpath_common/pull/251#issuecomment-3221160263.

Per the [ROS2 status page](https://repo.ros2.org/status_page/rolling_default.html?q=ros-jazzy-ewellix-description), the package made it to the building repo on the 21st, and the testing repo on the 25th, but is not yet in the main repo. Thus, any build script which transitively depends on the `git` version of clearpath_common, e.g. via clearpath_simulator or clearpath_robot, will fail if it checks the return value of rosdep.

**To Reproduce**

```
host$ docker run -it --rm ros:jazzy

docker$ mkdir -p /ws/src; cd /ws

docker$ cat >dependencies.repos <<EOF
repositories:
  clearpath_common:
    type: git
    url: https://github.com/clearpathrobotics/clearpath_common.git
    version: jazzy
EOF

docker$ vcs import src < dependencies.repos

docker$ rosep install -y --from-paths src
```

tonybaltovski, 2025-08-28:
It was [added](https://github.com/ros/rosdistro/pull/47325) on 19th to OSRF's buildfarm and indeed [building](https://build.ros2.org/job/Jbin_uN64__ewellix_description__ubuntu_noble_amd64__binary/).  However, it has not been synced from testing  to main.  Are you installing this from source as [2.7.0](https://github.com/ros/rosdistro/pull/47386) hasn't been merged into the OSRF's rosdistro. The sync to main happens once a month by the OSRF.  Can you add the [Clearpath package server](https://docs.clearpathrobotics.com/docs/ros/installation/offboard_pc#clearpath-packages) which has already synced it to main? OR you could switch to the [testing packages](https://docs.ros.org/en/humble/Installation/Testing.html)?  The is a bit of a niche case where a new package was added that did not exist before.  Additionally, a quick fix could be telling rosdep to proceed despite missing package using `rosdep install -y -r --from-paths src`.

vtomnet, 2025-08-28:
Ah, you are correct, it seems we were accidentally installing both from rosdistro and from source. Installing from rosdistro does work. Thank you for your time!

closed 2025-08-28 by vtomnet