clearpathrobotics/clearpath_common #260: missing collision tags for some links of Ridgeback (r100.urdf.xacro)

SyZbidi, 2025-09-09:
Hi,

while loading the robot description of the ridgeback for motion planning purposes, I noticed that the collision tags are not defined in all the links of the [robot.](https://github.com/clearpathrobotics/clearpath_common/blob/humble/clearpath_platform_description/urdf/r100/r100.urdf.xacro) the links below are then not considered as obstacles the planning scene.

is there a reason why only specific links have collision tags and the rest don't ?

thank you 

```
[moveit_robot_model.robot_model]: Link axle_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link front_rocker_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link rear_rocker_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link front_cover_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link front_lights_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link left_side_cover_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link rear_cover_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link rear_lights_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[moveit_robot_model.robot_model]: Link right_side_cover_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
```

SyZbidi, 2025-09-15:
I'm trying to use mobile-manipulator motion planning with ridgeback being the mobile base. the covers don't have collision geometry, so they aren't taking into account by the motion planner. 

> Those links had their collision tags removed since they are internal links and it reduces computation resources for MoveIt.

True, but front_cover_link, rear_cover_link, left_side_cover_link, and right_side_cover_link are external links. I can make a PR if you agree?

tonybaltovski, 2025-09-15:
They are indeed external but they are covered by the [chassis collision mesh](clearpath_platform_description/meshes/r100/body-collision.stl) (in-comparison with [chassis visual mesh](https://github.com/clearpathrobotics/clearpath_common/blob/jazzy/clearpath_platform_description/meshes/r100/body.stl).

SyZbidi, 2025-09-16:
ah I see, makes sense now. thank you for the explanation

[image]

closed 2025-09-16 by SyZbidi